IntergatedCircuits/USBDevice #1: Enumeration fails with gcc optimizations enabled

zzattack, 2018-06-26:
I've adopted this library on a STM32F042 project and ran into an issue where the device will not enumerate on the host because of a problem related to the device descriptor:
`Windows has stopped this device because it has reported problems. (Code 43)
A request for the USB device descriptor failed.`

Tracing the code that is executed, the device only gets one GetDescriptor request with value 0x01 for the device descriptor itself. Config or string descriptors are never requested.

When all gcc optimizations are disabled, the device does enumerate just fine and all my interfaces and endpoints function perfectly. GCC invoked either no optimization -O0 or  -Os, -O3 results in proper enumeration. With fewer optimizations -O1 or -Og enumeration fails.

IntergatedCircuits, 2018-06-26:
The most likely explanation is that the __packed keyword isn't correctly interpreted by the compiler. The easiest way is to check the contents of the Setup field. If e.g. Length > 0x100 then this is indeed the root cause.

zzattack, 2018-06-26:
This was my initial suspicion but it appears not to be the problem, __packed is preprocssed as __attribute__((packed)) as respected by gcc. My other guess is a missing volatile specifier somewhere.

zzattack, 2018-06-26:
Some more debugging reveals that the issue lies deeper. Without having touched the USB code, the interrupt handler now never sees a correct transfer (USB_ISTR_CTR set). The `USB_ISTR_RESET` and `USB_ISTR_SUSP` events do occur, but the followup `USB_ISTR_WKUP` never does when compiling with -Og or -O1. Instead the `USB_ISTR_RESET` happens a few more times before the host gives up.

IntergatedCircuits, 2018-06-26:
I can reproduce this on my F072Disco. Could you check if your USB.CNTR.FSUSP bit is stuck after the failed enumeration?

zzattack, 2018-06-26:
It is indeed. CNTR holds 0x9C08 while after correct enumeration the register holds value 0x9C00.

zzattack, 2018-06-26:
You're correct, clearing the SUSP bit before setting FSUSP  and LPMODE fixes the issue.

IntergatedCircuits, 2018-06-26:
I have pushed the correction to STM32_XPD, if this resolves your issue you can close it. Thanks for the finding.

zzattack, 2018-06-26:
It does indeed, thanks for the resolution.

closed 2018-06-26 by zzattack