IntergatedCircuits/USBDevice #2: Cannot differentiate between HID report types

zzattack, 2018-06-26:
In `hid_dataStage(...)` the (optional) report id is extracted and copied from the `dev->Setup.Value` LSB into the packet payload.
The MSB in `dev->Setup.Value` indicates whether this is a SetReport (0x02) or SetFeature (0x03). There seems to be no way to differentiate between the two in the SetReport callback.
I believe a similar mechanism is used for GetReport (0x01) and GetFeature (0x03) on the other endpoint.

IntergatedCircuits, 2018-06-26:
The HID standard allows all types of reports to be read or written through the control pipe with GetReport and SetReport requests. The MSB of the wValue field identifies the report type (1 - input, 2 - output, 3 - feature).

The current implementation is deliberately limited, it expects that only feature reports are communicated through the control pipe. Rationale:
1. It simplifies the application interface.
2. The IN and OUT reports should be in any case sent over the interrupt endpoints, therefore a secondary channel to accessing these isn't necessary.

zzattack, 2018-06-26:
As far as I'm aware it is valid for a report descriptor to have both a INPUT and FEATURE report with identical REPORT_ID. In this case, how will the user-defined interface be able to handle the feature reports it defined?

IntergatedCircuits, 2018-06-26:
It is of course a valid thing to do, but I chose to constrain the application in this regard, since it is the same application that defines the report descriptor as well.

zzattack, 2018-06-26:
Aha, that makes sense then. Thanks for clarifying.

closed 2018-06-26 by zzattack

zzattack, 2018-06-30:
I might be misunderstanding but is the USB stack supposed to not receive non-feature Output reports at all?
I'm experiencing some odd behavior that I cannot explain. When I define my interface without report id's, I can use CreateFile/WriteFile to write to my device and have .SetReport fire for both Feature reports and Output reports.
When I change my descriptor to have a report id, I can no longer get this to work. I'm aware of the filtering that Windows does when sending such reports, i.e. the report id must be the first byte, this report must actually be in the report descriptor, and you must send a packet with the length exactly equal to that of the longest defined report + 1 for the id. 

If I change the report descriptor from Output to Feature report, I can just HidD_SetFeature as expected.

What confuses me is that presence of report id's seem to matter here. With your rationale, I would expect the callback not to be invoked from either WriteFile call, but since it does when not using report id', I'd expect the same when using them.

To add to my confusion: I found a (closed-source) test utility (forcetest.exe) that consistently gets my .SetReport callback to be invoked, regardless of whether or not I'm using report id's. It does not use feature reports.

Could you clarify whether we're either not able to distinguish feature from non-feature reports, or whether we shouldn't be able to receive non-feature reports in the first place?