IntergatedCircuits/USBDevice #3: HardFault from USB_prvReadPMA

zzattack, 2018-06-27:
I have an very simple interface with IN and OUT endpoints, no report id's. As soon as I send some data from the host to this interface, due to accessing a nullptr in `pxEP->Transfer.Data`. 

I'll try to list as much relevant info as possible at the moment of failure:
* USB_prvReadPMA() called with pucDstBuf = 0x00000000, usPmaAddress = 0x00a8, usDataCount = 0x0008
* usEpId = 0x02; that's correct OUT EP for this interface
* pxEP is properly set, but it's TRANSFER structure might not be:
** Transfer | {Data = 0x00000000, Length = 0x0000, ...}
** MaxPacketSize | 0x0008 
** Type | USB_EP_TYPE_INTERRUPT 
** State | USB_EP_STATE_IDLE 
** IfNum | 0x01 '\x01' 
** RegId | 0x02 '\x02' 

Report descriptor and interface definitions below.
``` 
__ALIGN_BEGIN static const uint8_t hid_rpt_cfg[] __ALIGN_END = {
	0x06,	0x00,	0xFF,            // Usage Page = 0xFF00 (Vendor Defined Page 1)
	0x09,	0x01,                    // USAGE (Vendor Usage 1)
	0xa1,	0x01,                    // COLLECTION (Application)
	0x05,	0x01,                    //   USAGE_PAGE (Generic Desktop)
	0x15,	0x00,                    //     LOGICAL_MINIMUM (0)
	0x26,	0xff,	0x00,            //     LOGICAL_MAXIMUM (255)

	// write config command
	0x09,	0x00,                    //     USAGE (Undefined)
	0x95,	CFG_CMD_REPORT_SIZE,     //     REPORT_COUNT
	0x29,	CFG_CMD_REPORT_SIZE,     //     USAGE_MAXIMUM
	0x19,	0x01,                    //     USAGE_MINIMUM
	0x75,	0x08,                    //     REPORT_SIZE (8)
	0x91,	0x00,                    //     OUTPUT (Data,Ary,Abs)
	0x29,	CFG_CMD_REPORT_SIZE,     //     USAGE_MAXIMUM
	0x19,	0x01,                    //     USAGE_MINIMUM
	0x81,	0x00,                    //     INPUT (Data,Ary,Abs)

	0xc0                             // END_COLLECTION   
};

const USBD_HID_AppType cfgApp = {
	.Name = "sample config if",
	.Report = {
		.Desc = hid_rpt_cfg,
		.Length = sizeof(hid_rpt_cfg),
		.IDs = 0, // no report ids
	},
	.SetReport = Configurator_SetReport,
	.GetReport = Configurator_GetReport,
};

USBD_HID_IfHandleType hconfigurator_if = {
	.App = &cfgApp,
	.Config.InEp.Size = CFG_CMD_REPORT_SIZE,
	.Config.InEp.Interval = 20,
	.Config.OutEp.Size = CFG_CMD_REPORT_SIZE,
	.Config.OutEp.Interval = 20,
}, *const configurator_if = &hconfigurator_if;
```

IntergatedCircuits, 2018-06-28:
Each time you want to receive data through the OUT endpoint, you need to call USBD_HID_ReportOut(). Could you post the way you do this? If you aren't calling this function yet, and still be able to receive data through the endpoint, that's another issue I have to look into.

IntergatedCircuits, 2018-06-28:
There are two call paths for SetReport, only data received through the interrupt endpoint may cause a problem.

zzattack, 2018-06-28:
Sorry for the ninja-edit, the 'working' interface wasn't correctly defined either. Not sure if this changes your assessment.

IntergatedCircuits, 2018-06-28:
There is indeed a bug: the OUT endpoint should always be in NAK state unless the user has prepared the necessary reception buffer. This isn't the case for EP0 as it always uses the CtrlData buffer.

IntergatedCircuits, 2018-06-28:
I pushed an update to STM32_XPD, it should fix this error. Again, thanks for the catch.