IntergatedCircuits/USBDevice #5: Hotplugging results in failed enumeration/invalid device descriptor

zzattack, 2018-09-30:
I built a device that is sometimes powered through USB and sometimes externally. When powered over USB this issue is never observed, but when the device is already running and then I plug in the USB cable, about 1/3 times Windows fails to enumerate the device with message "invalid device descriptor".
Is there anything special we should do to support hotplugging?

IntergatedCircuits, 2018-09-30:
There is no known reason for this to happen, the host always starts USB communication with a reset event which puts the device in a well-defined initial state, and any data exchange happens afterwards. I have never noticed this issue with the same setup. I suggest you record the USB packets on either host or device side and check what goes wrong.

zzattack, 2018-09-30:
I tried some logging with Device Monitoring Studio but very little comes up for the hotplugged device. 
There's 3 events 'Device connected', 'Surprise removal' and 'Device disconnected' quickly following each other. I'll try to add some logging on the device side but in my experience, the additional overhead from logging has been cause to failed enumeration for me. Do you have any tips for this?

IntergatedCircuits, 2018-09-30:
This sounds like a physical connection issue causes line transitions during the hot-plugging. I'd try different cables or connectors to see if it's hardware specific.

On the firmware side you may call the `USBD_Connect()` only after USB host power has been detected by the device. Although this isn't a reliable method since the time between the power pins contacting and the data pins contacting can vary greatly depending on insertion speed.

zzattack, 2018-09-30:
Different combinations of cables and USB hubs doesn't seem to affect the issue. 
In my case the design is limited as the microprocessor doesn't have any indication whether or not it is being fed USB power. I've in the past built a similar device which used the microchip USB stack where this never formed an issue. Perhaps I can find out if the firmware can recover from this situation if it occurs.

zzattack, 2018-09-30:
I haven't specified the .Deinit callback member for my interfaces so there's certainly no blocking call there.

What I did just notice is that the `USBD_ResetCallback` function in `usbd.c` is executed every few seconds and there's some attempt at exchanging descriptors.
There's a single execution of `USBD_DevRequest` for a `USB_REQ_GET_DESCRIPTOR` but it seems the `dev->Setup.Request` may read 0x0000 or my debugger isn't helpful.

zzattack, 2018-09-30:
This is what a successful enumeration looks like for me.
(rt=dev->Setup.RequestType.Type, rq=dev->Setup.Request, sv=dev->Setup.Value)

> [USB] USBD_ResetCallback()
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback()
[USB] USBD_DevRequest() rt=0 rq=5
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=f00
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=f00
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=330
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=300
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=320
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=9
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=301
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=301
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=302
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=302
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=330
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=330

Failed enumerations look like this:
>[USB] USBD_ResetCallback()
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback()
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback()
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback()
[USB] USBD_DevRequest() rt=0 rq=6
[USB] USBD_GetDescriptor(), sv=100

Right after the 3rd device descriptor request, Windows shows the 'invalid device descriptor' message.

IntergatedCircuits, 2018-09-30:
Could you also log the state of USB_ISTR:PMAOVR and USB_ISTR:ERR register bits?

zzattack, 2018-09-30:
Both during successful and failed enumeration they read 0 at each step.

IntergatedCircuits, 2018-09-30:
Then it's time for a logic analyzer, otherwise I'm out of ideas.

zzattack, 2018-09-30:
When I'm logging the bits by reading them as `USB->ISTR.b.ERR` and `USB->ISTR.b.PMAOVR` they show up as 0, but when I log the unformatted ISTR value it actually reads 0x2300. 

Here's a successful enumeration:
> [USB] USBD_ResetCallback(), ISTR:0100
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback(), ISTR:0300
[USB] USBD_DevRequest() rt=0 rq=5, ISTR:0300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=f00
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=f00
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=330
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=320
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=9, ISTR:0300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=301
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=301
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=302
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=302
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=330
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=330
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=302
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=301
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:0300
[USB] USBD_GetDescriptor(), sv=200

Here's one that was unsucessful:
[USB] USBD_ResetCallback(), ISTR:2300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:2300
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback(), ISTR:2300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:2300
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback(), ISTR:2300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:2300
[USB] USBD_GetDescriptor(), sv=100
[USB] USBD_ResetCallback(), ISTR:2300
[USB] USBD_DevRequest() rt=0 rq=6, ISTR:2300
[USB] USBD_GetDescriptor(), sv=100

About half the time though, ISTR still reads 0x0300 and fails in otherwise identical fashion.

IntergatedCircuits, 2018-10-15:
Any progress on this on your side? I still believe the error stems on the physical layer. You could cross-check it by flashing an STM32Cube-generated USB project, and if the problem persists, turn to their forum for help. There's definitely a larger audience.

zzattack, 2018-10-15:
Haven't been able to put in the effort yet but I did try generating a CubeMX simple HID project to repeat the tests. I've unplugged/replugged the USB cable 100x while the device was already being powered. The devices enumerated 100x without failure so I do believe it's a software issue.
I'll try repeating the tests with a bare skeleton using your library tomorrow.

IntergatedCircuits, 2019-01-17:
In the meantime I have made glue code for STM32Cube, this could help you reduce the effort of creating a bare USBD project:
https://github.com/IntergatedCircuits/USBDevice4Cube

zzattack, 2019-03-16:
I have updated the usb-related source files and the program seems to have gone away. After some pinpointing, it seems the latest changes to xpd_usb.c [here](https://github.com/IntergatedCircuits/USBDevice4Cube/commit/73dfb40c689590221c4bfa012b0b9596798bc742#diff-e9f59d72c47d2541349380d62e7abaaf) fixed my issue.

One thing I wonder about is the macro definition of `USB_LPMCSR_LMPEN` should actually be `USB_LPMCSR_LPMEN` as there may be a mismatch there between some of the files I ended up with. May be a mixup on my end, but LPM makes more sense to me given what the abbreviation stands for.

IntergatedCircuits, 2019-03-16:
Thanks for coming back on this, I'm glad to hear that the problem is resolved.

Indeed the LPMEN bit definition is screwed up in the CMSIS header, I know this because in the reference manual it is correctly referenced as LPMEN (which stands for **L**ink **P**ower **M**anagement **En**able btw.). It is redefined in the wrapper project [here](https://github.com/IntergatedCircuits/USBDevice4Cube/blob/24826257ed3afced00ce2cfa7fd8e90ea188946e/XPD_USB/xpd_config.h#L51), so if you are missing this then there will be a compilation error.

closed 2019-03-16 by IntergatedCircuits