IntergatedCircuits/USBDevice #8: USB CDC transmission problem/solution when port is not opened

VaZso, 2019-01-11:
Dear Benedek,

First of all, thank you for developing and sharing your project, I think it is a much better solution compared to STM32 original implementation.

I am using an STM32F407 microcontroller using Eclipse and STM32 HAL implementation.
Thanks to descriptions, I have successfully managed to compile and run USBDevice in a test project.

As a starting point, I have used VCP_IF interface from your project and simplified it a bit.

My host computer is running Linux.

The main program calls transmit function in every millisecond and sends a message using VCP interface to host computer.

In that case, STM32 "hangs" after calling USBD_CDC_Transmit() function after a few times if the port is not open or the cable was reconnected.

About STM32 hangs... it is actually running, but it constantly calls USB_vIRQHandler() routine. If I unplug then replug the cable, the main program starts to run again till a call of USBD_CDC_Transmit() function occurs, then it hangs again by continuously calling USB_vIRQHandler() function.
I don't know the exact reason of the loop, but calling USBD_Disconnect() then USBD_Connect() function solves the situation, communication works again.

So, to solve it, when (USBD_HandleType).Suspend callback occurs, I am calling USBD_Disconnect() and USBD_Connect() once so system calls it every time I disconnect USB cable (anyway, it is usually also fired up on cable reconnection). Currently, a communication delay of 1000ms is also applied here.

It solves the problem which happens after cable reconnection, but STM32 still halts if I wait too long to open the serial port after plugged in USB cable.

At the time of Close callback (in USBD_CDC_AppType) is called, I have also added an option for the software to discard calling USBD_CDC_Transmit() till Open callback not called or at least for 100ms.

It helped a bit, but STM32 still hangs if it wants to send data before the serial port was opened at the host computer.

If I open the port and close it at host computer in time, STM32 does not hang and I can reopen serial port again.

So,  I have started to find a way to determine if the port was opened after Close/Open callbacks fired up.

I have found a similar problem <a href=https://community.st.com/s/question/0D50X00009XkgdmSAB/usb-cdc-class-how-to-determine-if-vcp-has-been-opened>here</a>.
Based on that, I have modified usbd_cdc.c and included a callback at "CDC_REQ_SET_CONTROL_LINE_STATE" stage.
In VCP function, the software continues to discard calling USBD_CDC_Transmit() function after Close callback happens while ControlLineState callback is not fired up (it happens when I open serial ttyACM interface).

After this modification, STM32 seems to run well even after disconnect/connect the cable and also if the serial port was not opened. If I open it, my messages are displayed well.

My modified callback structure is below:

`/** @brief CDC application structure */
typedef struct
{
    const char* Name;   /*!< String description of the application */

    void (*Open)        (void* itf,
                         USBD_CDC_LineCodingType* coding);  /*!< Open port */

    void (*Close)       (void* itf);        /*!< Close port */

    void (*Received)    (void* itf,
                         uint8_t* data,
                         uint16_t length);  /*!< Received data available */

    void (*Transmitted) (void* itf,
                         uint8_t* data,
                         uint16_t length);  /*!< Transmission of data completed */

    void (*ControlLineState)	(void* itf);        /*!< Control line state changed */

#if (USBD_CDC_BREAK_SUPPORT == 1)
    void (*Break)       (void* itf,
                         uint16_t len_ms);  /*!< Interface interaction through the control channel */
#endif /* USBD_CDC_BREAK_SUPPORT */
}USBD_CDC_AppType;
`

My modified  cdc_setupStage function is below:
`/**
 * @brief Performs the interface-specific setup request handling.
 * @param itf: reference of the CDC interface
 * @return OK if the setup request is accepted, INVALID otherwise
 */

/** @brief USB setup request */
static USBD_ReturnType cdc_setupStage(USBD_CDC_IfHandleType *itf)
{
    USBD_ReturnType retval = USBD_E_INVALID;
    USBD_HandleType *dev = itf->Base.Device;

    switch (dev->Setup.RequestType.Type)
    {
        case USB_REQ_TYPE_CLASS:
        {
            switch (dev->Setup.Request)
            {
                case CDC_REQ_SET_LINE_CODING:
                    /* Reset the data interface */
                    cdc_deinit(itf);

                    retval = USBD_CtrlReceiveData(dev, &itf->LineCoding);
                    break;

                case CDC_REQ_GET_LINE_CODING:
                    retval = USBD_CtrlSendData(dev,
                            &itf->LineCoding, sizeof(itf->LineCoding));
                    break;

                case CDC_REQ_SET_CONTROL_LINE_STATE:
			USBD_SAFE_CALLBACK(CDC_APP(itf)->ControlLineState, itf);

                    retval = USBD_E_OK;
                    break;

#if (USBD_CDC_BREAK_SUPPORT == 1)
                case CDC_REQ_SEND_BREAK:
                    /* Simply pass the request with wValue */
                    if (CDC_APP(itf)->Break != NULL)
                    {
                        CDC_APP(itf)->Break(itf, dev->Setup.Value);
                        retval = USBD_E_OK;
                    }
                    break;
#endif /* USBD_CDC_BREAK_SUPPORT */

                default:
                    break;
            }
            break;
        }

        default:
            break;
    }

    return retval;
}
`
Another small modification of the code was in USBD_CDC_Notify function, where "notice->Header->Length" should be "notice->Header.Length":

`#if (USBD_CDC_NOTEP_USED == 1)
/**
 * @brief Sends a device notification to the host.
 * @param itf: reference of the CDC interface
 * @param notice: pointer to the notification message to send
 * @return BUSY if the previous transfer is still ongoing, OK if successful
 */
USBD_ReturnType USBD_CDC_Notify(USBD_CDC_IfHandleType *itf, USBD_CDC_NotifyMessageType *notice)
{
    USBD_ReturnType retval = USBD_E_ERROR;

    if (itf->Config.NotEpNum < USBD_MAX_EP_COUNT)
    {
        uint16_t length = sizeof(notice->Header) + notice->Header.Length;
        retval = USBD_EpSend(itf->Base.Device, itf->Config.NotEpNum, notice, length);
    }

    return retval;
}
#endif`

Could you please update your code to include this (or similar) callback function?

Thank you in advance and thank you again for your work.

Best Regards,
Zsolt

IntergatedCircuits, 2019-01-12:
Hi Zsolt, welcome to Github!

Let me start with some advice on how to use this platform:
1. Report each problem in an individual issue. That way we can track the status and close each of them when they are solved. The syntax error is now fixed under this issue number, so let's only keep discussing the rest.
2. In case you come up with the solution to a problem, you can also create a pull request. (Fork this repo, push the changes on your branch, then open a pull request.)

Now, onto the specific issue. The CDC function doesn't open the endpoints of its data interface until a valid line coding is set by the host. The `USBD_EpSend()` method and therefore the `USBD_CDC_Transmit()` should return USBD_E_BUSY without any further action when the host didn't open the port yet. So what you are reporting is unexpected to me. I would need some more details on how your implementation works and some debugging info as well:
a) I guess the USBD stack is running in the default interrupt context, while you have a main program in thread context sending CDC data, is that correct?
b) What is the return value of `USBD_CDC_Transmit()`, and what is the value of the CDC Tx `USBD_EpHandleType` when the hang is occurring?
c) What are the circumstances of the halt? I understand that your design is self-powered. Does it halt the first time when the system starts and the port isn't open, or does the halt occur only after the port has been opened, but there was a cable replug, or port closing?
d) It would also be useful if you could debug which USB interrupt request signal keeps firing when the halt occurs.

On the topic of using Set_Control_Line_State request with DTR bit set to detect an opened port:
The forum that you linked in turn links to this [discussion](https://stackoverflow.com/questions/5338875/detecting-open-pc-com-port-from-usb-virtual-com-port-device), where it is also mentioned that this solution can only work with certain applications. I checked this myself, and the .NET SerialPort implementation for example doesn't set this by default. Therefore I'm not in favor of changing the current mechanism - since the line coding parameters have to be sent by the host, it is the best known solution to trigger `Open` when that happens.

VaZso, 2019-01-13:
Hi,

Thank you your reply.

I have tried to look deeper in, however, currently I feel myself a bit tired caused by this thing (even it is late now).

Now I have removed waiting and checkings in my vcp_if app code (LineCoding.DataBits was also checked).

Currently, the main program is an infinite while cycle which reads HAL_GetTick() function to get system timer's value and calls VCP_Transmit() function which now directly calls USBD_CDC_Transmit() function when timer value is increased.
It also inverts some LEDs (by 500ms increase of timer) and checks printf redirection's buffer and enables UART interrupt if it has new data.

USB initialization code comes from USBDevice4Cube's hal_usb.c file.
Interrupt calls OTG_FS_IRQHandler() defined here which calls USB_vIRQHandler().

So now, when STM32 starts, after initialization, USBD_CDC_Transmit() function is immediately called about once per a millisecond.

If USB cable is not attached, USBD_CDC_Transmit() result code is always 2 (USBD_E_BUSY) and my LEDs are blinking.
If I attach USB cable and open the connection, I see one of my messages displayed (but only one).
At my serial console, I see result code of 2, USB Close callback, USB Open callback and result code of 0 (USBD_E_OK) as last messages and my LEDs are stop blinking.

The same happens if I power the STM32 directly from my USB cable.

At this point, if I take a breakpoint at USB_vIRQHandler(), the following register values can be read:
GINTSTS: 1000100100001001000010000111000
DAINT: 10
DIEPINT1 10000010000000  (TXFE / NAK interrupt)

It looks exactly like in the discussion you linked.

If I run the same code on STM32, but include a delay before the while() cycle and I open the port before this delay is elapsed (it is 5 seconds now), then I can see all of the messages sent by STM32 while my serial console shows USB Closed, USB Opened and USBD_CDC_Transmit() result codes of 0.
This time VCP works without problems. If I close serial port, USBD_CDC_Transmit() result code is 2, if I open it, result code is 0 and STM32 messages coming again and my LEDs are blinking as they should.
The same happens if I power STM32 directly from USB cable or from external power.

At this point, if I unplug USB cable, software still works well, result code is 2 and LEDs are blinking.
Then, when I replug USB cable and open the port, I see only one message of STM32 communication, LEDs are not blinking and on serial console, I currently see USB Opened, USB Closed, result code 0, USB Opened, result code 0, USB Closed, result code 0, USB Opened as last messages, where result code is the result of USBD_CDC_Transmit().

Now I can see the exact same values of GINTSTS, DAINT and DIEPINT[1].
In USBD_EpHandleType, I can see a message data, a length of 20 bytes and Progress of 20.

As of the ControlLineState callback, I think you may put it in the software in case of somebody need it even if it is not getting closer to a real solution of the problem above.
It is only a few lines of code, does not brake anything and can be also switchable by a define.

Sorry if I missed something, I have tried to write everything in an appropriate way but it may happen I have to clarify it.

The best solution would be if the hang of USB system could be eliminated.

What is your suggestion for me to try?

If I understand well, USBD_CDC_Transmit() should not do any problem even if it is called when port is still not properly enumerated by host or just only not opened yet.
Am I right?

Should it also survive cable disconnection / reconnection event or does it need something like calling USBD_Disconnect() / USBD_Connect() functions?

Thank you your reply in advance.
Regards,
Zsolt

IntergatedCircuits, 2019-01-13:
I added the option to get the SetControlLineState callback from the CDC function.

You could check one more thing for me: what are the line coding parameters that the application receives when CDC is opened? Are these parameters what you configured when you open the port, or some system default?

As for the hang, I will have to check this myself later, I will update this issue wiwth the results when I have them.

VaZso, 2019-01-13:
Thank you.

This is some system default.

In .Open callback, I have printed out (through my UART line) CharFormat, DTERate, DataBits and ParityType, respectively.

That way, when I connect it, I see:
USB Opened (0, 9600, 8, 0)
USB Closed
USB Opened (0, 57600, 8, 0)

If I set a baudrate on host, I can see it changes to the value set:
USB Closed
USB Opened (0, 115200, 8, 0)
USB Closed
USB Opened (0, 921000, 8, 0)

While I was switching speed, communication worked well.
However,my LEDs are stopped working when I switched / set back baud rate from a non-standard value like switching from 921000 to 9600 (I don't know if the non-standard value has to do anything with it).

After that, I still see messages from USB Closed / Opened callbacks with the appropriate values set by host system. When I change baud rate, message is displayed by UART and my LEDs states are change once.

So, I can see open callbacks through UART line even if my LEDs are not blinking (__HAL_UART_ENABLE_IT is called when a character was put in buffer).

Later, I was switching between standard baudrate values and my LEDs are also stopped blinking.
Now I see the following values in registers:
GINTSTS: 1000100100001001000010000101000
DAINT: 10
DIEPINT[1]: 10000010000000

Anyway, I think usb-modeswitch functionality is still working on host computer even though I have disabled and removed this package. It's aim is to switch USB modems from "CD / driver" mode to modem mode and it tries to send messages to ttyACM (CDC-ACM) devices a few seconds after connection if the port is not opened yet.

VaZso, 2019-01-14:
For testing / reproduction:
The problem only occurs if a call of USBD_CDC_Transmit() happens in the "not appropriate" time.

If STM32 only wants to send data when I press a button, then everything seems to work well even after USB reconnection. (...but I don't press it when I connect the system.)

So, constant data sending while USB connect/disconnect or even while changing line coding parameters could trigger this behaviour.

My goal is to have a CDC-MSC + CDC-ACM interface where the latter is to provide information of the current state of the device to see what's happening in the background.
That is why I have tried continuous sending, as in the case above the system would provide information anytime which only needed if somebody opens the ACM device.

IntergatedCircuits, 2019-01-14:
Did you try the `console_if` from the Templates? If you need the CDC for a variation of `printf()` style logging, you should try it first.

VaZso, 2019-01-14:
Do you think it helps?

Anyway, I currently have a printf redirection to USART1 and CDC-ACM would not necessarily a printf redirection.

I assume it may have a similar behaviour if data is sent "too often".
I may give it a try but it would be good to know what is the source of this thing.

VaZso, 2019-01-14:
Now I have tried to use console_if.c template, but still without success.

I kept everything the same except skipped loading vcp interface and included console_if plus disabled my printf redirection.

At frist, host said "bad CDC descriptors" twice when STM32 connection was made.
Then I have removed ".Config.Protocol = 0xFF", line from "USBD_CDC_IfHandleType _console_if" (also tried with ".Config.Protocol = 0x01"). That way host had been identified it as ttyACM0 device, which is okay.

I could open the port, but nothing was displayed (printf redirection did not work).
.Open callback was called as it should. In console_if_send, Length value is 4 and USBD_CDC_Transmit() seems to be only have \0 characters in buffer.
Still haven't digged in it deeper.

I have put a USBD_CDC_Transmit() function to execute every 500ms in main cycle, but I also had to include some delay before and after it to get it displayed at host computer.
(Maybe because of other printf calls...)
When I have opened the port several times, I was also able to reach the same behaviour as earlier, but I understand this data was not sent by console_if itself.

Currently I'm stuck here.

VaZso, 2019-01-15:
One interesting thing is - as far as I can see - when communication is working, in USB_prvTransmitPacket() function, the value of ulFifoSpace is 64 which is the same value of pxEP->MaxPacketSize.
So, the value of DTXFSTS is 16.

When my problem occurs, the value of DTXFSTS is 11, thus, ulFifoSpace is 44 which is less than 64.
So, USB_prvWriteFifo() is not called.

As pxEP->Transfer.Progress is still greater than 0 (currently full of 19 bytes), it calls SET_BIT(pxUSB->Inst->DIEPEMPMSK, ulEpFlag);

When program calls again USB_prvTransmitPacket() function, the value of DTXFSTS is still 11 and it does the same.

IntergatedCircuits, 2019-01-16:
That's a good way to start. Can you check how it works if you flush all FIFOs at the Reset signal instead of only Tx0? So replace
```C
        /* Handle Reset Interrupt */
        if ((ulGINT & USB_OTG_GINTSTS_USBRST) != 0)
        {
            /* Clear IT flag */
            USB_FLAG_CLEAR(pxUSB, USBRST);

            pxUSB->LinkState = USB_LINK_STATE_ACTIVE;

            /* Stop any ongoing Remote Wakeup signaling and EP0 transfers */
            USB_REG_BIT(pxUSB,DCTL,RWUSIG) = 0;
            USB_prvFlushTxFifo(pxUSB, 0);

            /* Clear EP interrupt flags */
            USB_prvClearEpInts(pxUSB);

            /* Set default address (0) */
            USB_vSetAddress(pxUSB, 0);
}
```
with this:
```C
        /* Handle Reset Interrupt */
        if ((ulGINT & USB_OTG_GINTSTS_USBRST) != 0)
        {
            /* Clear IT flag */
            USB_FLAG_CLEAR(pxUSB, USBRST);

            pxUSB->LinkState = USB_LINK_STATE_ACTIVE;

            /* Stop any ongoing Remote Wakeup signaling */
            USB_REG_BIT(pxUSB,DCTL,RWUSIG) = 0;

            /* Flush the FIFOs */
            USB_prvFlushTxFifo(pxUSB, USB_ALL_TX_FIFOS);
            USB_prvFlushRxFifo(pxUSB);

            /* Clear EP interrupt flags */
            USB_prvClearEpInts(pxUSB);

            /* Set default address (0) */
            USB_vSetAddress(pxUSB, 0);
}
```

IntergatedCircuits, 2019-01-16:
The other thing that you can try instead is to add this line to `USB_vEpClose()`:
`USB_prvFlushTxFifo(pxUSB, ucEpNum);`

VaZso, 2019-01-17:
I have tried your first suggestion, but behaviour was the same - I have not tested what is happening, but my LEDs are halted.

After reverted this modification and put USB_prvFlushTxFifo(pxUSB, ucEpNum) call in USB_vEpClose() function, the problem was **disappeared**. :-)

So far I see STM32 messages every time I open serial port, change baud rate or even disconnect/reconnect USB cable several times.
I have also removed initial delay I put earlier before main while() cycle because a hang was occurred when STM32 sent data continuously before the port was opened (mentioned above earlier).
Now it seems STM32 is running well regardless what I am trying to do with USB CDC communication.

Naturally, this continuous data sending was a test and not a real-life behaviour, but I wanted to eliminate the chance of having similar hang 'sometimes' in any applications I use CDC-ACM.

So, thank you very much your effort and your help, your latter suggestion is **solved** this problem.
Now it works perfectly well.

IntergatedCircuits, 2019-01-17:
Thanks for the verification, I will push the fix to the proper repos and close this ticket afterwards.

closed 2019-01-20 by IntergatedCircuits