IntergatedCircuits/USBDevice #9: MSC configuration

VaZso, 2019-01-28:
Hello,

Could you please help me how to configure an MSC device properly?

I hope you may have something which may be used as example configuration showing basic requirements of device setup.
I think I miss some configuration to define or doing it completely wrong as my MCU generates HardFault.

So, if you may have something which can be used for proper device initialization, I would be happy.

Thank you your help in advance.

IntergatedCircuits, 2019-01-29:
As long as all the pointers are set correctly in your application, there shouldn't be any HardFaults caused by the library. In this case please post it as a ticket with relevant stack trace and a way to reproduce.

According to the documentation of `USBD_MSC_MountInterface()` the The interface reference shall have its `USBD_MSC_IfHandleType::Config` structure and `USBD_MSC_IfHandleType::LUs` reference properly set before this function is called.

The `USBD_MSC_LUType` - where LU stands for Logical Unit, a single piece of block storage - defines the mass storage API, with calls identical to other block file systems, e.g. **FatFS**. Its `Status` field should point to a struct that has up-to-date information about the block storage's state. The `Inquiry` field isn't that obvious, so here's an example:
```C
const USBD_SCSI_StdInquiryType inquiry = {
    .PeriphType = SCSI_PERIPH_SBC_2,
    .RMB = 1,
    .Version = 2,
    .RespDataFormat = 2,
    .AddLength = sizeof(USBD_SCSI_StdInquiryType) - 4,
    .VendorId = "IC",
    .ProductId = "Mass Storage",
    .VersionId = "1.00"
};
```
Keep in mind that the string lengths are fixed in this struct.

MarcoPio, 2021-07-27:
I am trying to get a MSC + CDC composite that must work against MS Windows. Currently the device is enumerated and some descriptors are read (I have an USB analyzer) the VCP is detected I see on bus just after set configuration some line coding commands. However the MSC is not started (usually a "get max LUN" I saw on other devices..,)
I used the msc_if from VaZso (Thank you) and my UsbDevice_Init looks like:

`/* Initialize the device */
    USBD_Init(UsbDevice, dev_cfg);

    /* All fields of Config have to be properly set up */
    msc_if->Config.InEpNum = 0x81;
    msc_if->Config.OutEpNum = 0x01;
    msc_if->Config.MaxLUN = 0;

    /* Mount the interfaces to the device */
    USBD_MSC_MountInterface(msc_if, UsbDevice);

    console_if->Config.InEpNum  = 0x82;
    console_if->Config.OutEpNum = 0x02;
    console_if->Config.NotEpNum = 0x83;

    /* Mount the interfaces to the device */
    USBD_CDC_MountInterface(console_if, UsbDevice);
` 
shortly on the bus i have :
get device descriptor  with device class = defined in interface (0)
then
get config descriptor where
3 interfaces
itf 0 = class mass storage (2 ep)
itf 1 = class cdc and control (1 ep)
itf 2 = class cdc data (2 ep)

So can you give some suggestion or working example?

benedekkupper, 2021-07-28:
I don't have a public project with MSC, but others have successfully integrated it, e.g. https://github.com/Smoothieware/SmoothieV2/blob/master/Firmware/Hal/usb/msc_if.c
Make sure that msc_if->Base.AltCount = 1 and msc_if->LUs are also properly set.

MarcoPio, 2021-07-29:
I have msc_if->Base.AltCount = 1 and msc_if->LUs properly setted.
I saw the Smoothie project and they have VCP+DFU or MSC stand alone.
I used their hdev_cfg values and this was sufficient to making appear MSC along VCP on Windows.
What I understood is that if VID and PID are 0x483 , 0x5740 from ST then the composite device is not recognized by Windows.
Changing VID to a different number worked for me.
Do you have any suggestion for VID and PID values to use as general rule ?