IntergatedCircuits/USBDevice #10: Audio class

XGudron, 2019-03-30:
Is audio class support planned?

IntergatedCircuits, 2019-03-30:
While it would be certainly nice to have, I don't have any projects planned at the moment that would require an Audio class interface, so there is no timeline for it.

XGudron, 2019-03-31:
I am developing a open-source project of DDC/SDR Radio, and I plan to make sound output through USB. Also, now the project has 2 virtual COM ports for debugging output and control from a computer. I tried to make an audio class by analogy with the CDC, but my knowledge is clearly not enough for this.
Can you advise the literature that will help me form the descriptors correctly?

IntergatedCircuits, 2019-03-31:
As far as I have seen, there are two important Audio class documents on the USB implementers forum: the latest official full-blown [audio class spec](https://www.usb.org/document-library/usb-audio-devices-rev-30-and-adopters-agreement) and a [simplified spec](https://www.usb.org/document-library/audio-device-class-spec-basic-audio-devices-v10-and-adopters-agreement) for the most common use-cases. I'd recommend that you check the latter. You can also get inspiration from other open-source projects, such as [Mbed](https://github.com/ARMmbed/mbed-os/blob/master/features/unsupported/USBDevice/USBAudio/USBAudio.cpp).

As a starting point I recommend that you check how the NCM implementation initializes the endpoints, as the audio class has the same mechanism, namely that the interface-specific endpoints are only opened once the alternate setting is set to the non-default 1. Another thing to bear in mind is that isochronous transfers aren't tested yet, so you might run into problems on the driver level. If that happens, document as much as you can from the circumstances, and open an issue.

XGudron, 2019-04-01:
Thank you for the information provided!
Descriptors are successfully registered and the device is detected.
But the safe transfer does not occur, the data is not transmitted.

class file https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/master/STM32/USBDevice/Class/AUDIO/usbd_audio.c
header file https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/master/STM32/USBDevice/Include/usbd_audio.h

errors on usb sniffer: https://yadi.sk/i/8UNzux8ygufiig

IntergatedCircuits, 2019-04-09:
Can you provide a permanent link to the source code? The current links are broken.

XGudron, 2019-04-09:
https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/1.4/STM32/USBDevice/Class/AUDIO/usbd_audio.c

https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/1.4/STM32/USBDevice/Include/usbd_audio.h

I could not make isochronous transfers, I had to rewrite the standard solution from stm32cube to the composite structure (CDC+CDC+Audio).
https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/master/STM32/Src/usbd_ua3reo.c

IntergatedCircuits, 2019-04-09:
So, the biggest issue with the code is these lines:
```C
USBD_ReturnType ret=USBD_EpReceive(pdev->Base.Device, pdev->Config.InEpNum, haudio->buffer, AUDIO_OUT_PACKET);
```
The directions in the USB context are observed from the Host point of view, so the IN direction means from Device to Host. To receive data on the device side, you have to use the OUT endpoints. These calls should use the OutEpNum parameter, which should be in the 1..15 range (which is further limited by hardware).

XGudron, 2019-04-09:
Exactly, thank you! my mistake...

WardBenjamin, 2020-12-17:
I would definitely be interested in using this library if an audio class was added - I'm working on some open-source audio hardware (equalizer, DAC, etc) where having several interfaces would be very useful for debugging.