IntergatedCircuits/USBDevice #11: console_if.c won't transmit on STM32F072 (CubeMX)

bentwire, 2019-04-25:
For some reason I can't get anything to send with console_if.c.  I get the device detected and it shows up as ttyACM0 on my linux system, but when I use minicom to read it I don't see any output from the device.  In my main.c in the while loop I am calling _write() directly so I would expect to see something but nothing is coming through.  

Not sure what I am doing wrong, probably something simple, do you have any suggestions?

Thanks for any help,
Mike

IntergatedCircuits, 2019-04-26:
First of all, the console_if that you were using is buggy, I pushed a fix that works.

Not receiving any data at all is something that shouldn't happen though, so there's something missing on your side. The best shot you have is to debug the code to see if `_write()`, `console_if_send()` and `USBD_CDC_Transmit()` are all called.

bentwire, 2019-04-26:
Your fix fixed it for me.  I didn't have to change anything else.  Thanks!

closed 2019-04-26 by bentwire