IntergatedCircuits/USBDevice #13: PC device using USB

nelsonhf, 2019-06-26:
My company is starting to work on a USB printer simulator that will connect to a PC via USB.

The simulator will be hosted in a PC, using Windows 10.

Do you think your library would be appropriate for our project? I mean both technically and legally. This device will be used exclusively by the company for testing our product so we will not make a direct profit from it.

Thanks

IntergatedCircuits, 2019-06-27:
Please check the [license](https://github.com/IntergatedCircuits/USBDevice/blob/master/LICENSE) for the legal requirements.
I cannot tell if this library is what you need from a technical perspective, since I don't have sufficient technical details to evaluate it. Do I understand correctly that you want to emulate a PC as a USB device? I don't see how that would be possible without specialized hardware.

nelsonhf, 2019-06-27:
Yes. I want a PC to behave as if it was a device.

I thought it would be possible because it wouldn't be too different from what is done to transfer files from PC to PC using a "USB transfer cable". Of course, we would use the same (or similar) cable in our project. Do you think I would need some hardware in addition to the special cable?

Thanks for the help

IntergatedCircuits, 2019-06-28:
Can you link me such a cable? I don't really see how it was possible without an embedded integrated circuit in the cable, as the USB devices have to report their properties to the host, an the host assigns the appropriate device driver based on that.

nelsonhf, 2019-06-28:
Here are 3 links from Amazon (I have them handy because I asked my company to buy them for evaluation):
  -- Plugable Transfer Cable Compatible ($18.55; 4.3 stars; 72-12-5-3-8/1233): https://www.amazon.com/Plugable-Transfer-Compatible-Computer-Software/dp/B005OTPVMY/ref=sr_1_6
  -- iLuv [...] ]High Speed USB File Transfer Cable ($9.99; 3.8 stars; 58-16-5-0-21/19): https://www.amazon.com/iLuv-Winning-Transfer-Shielded-Required/dp/B007PFZE9U/ref=sr_1_6
  -- Dell 6 ft. Easy Transfer USB Cable ($9.02; 3.8 stars; 45-11-22-0-22/9): https://www.amazon.com/Dell-Transfer-Cable-Windows-RJ57K/dp/B00LFQZE68/ref=sr_1_12 

Well... technically, they are not simply "cables" as they do have some circuitry in the middle but that should do a little bit more than "impedance matching", i.e., should not change the signal itself.

IntergatedCircuits, 2019-06-28:
I suggest you do some research on how USB works, because you have the wrong idea about it now. These cables have an IC inside them, that presents the cable as a device towards both PC hosts. It should be obvious from the price tag, compared to a simple USB-A to USB-A cable.

closed 2019-07-03 by IntergatedCircuits