IntergatedCircuits/USBDevice #15: CDC enumeration fails in Linux

LACoopConsulting, 2019-11-07:
When I plug my device into a Windows PC, it enumerates and all is happy. Unfortunately when I plug this same device into a PC running Ubuntu Linux 18.01 or a Beagle Board Black running Debian, I get this error:

`cdc_acm: probe of 1-1:1.0 failed with error -22`

Usually it's Windows that doesn't play nicely! Any thoughts?

IntergatedCircuits, 2019-11-07:
Unlike Windows, Linux requires the CDC interface to have a valid notification endpoint, so make sure you are using that precompile switch, and that the notif ep is valid on the device.

LACoopConsulting, 2019-11-07:
Worked like a charm, thanks!

closed 2019-11-07 by LACoopConsulting

dognotdog, 2020-04-01:
Just wanted to add that OS X behaves the same, you need the notification endpoint for the VCP to be properly detected. Maybe putting a note in the docs/code to this effect would do the trick? Or just making it non-optional? Is there any downside to it?

IntergatedCircuits, 2020-04-02:
Well it's a waste of an endpoint, since it's virtually never used, and endpoints are a bottleneck when you're designing composite devices (e.g. on STM32F407 there are only 3 non-control endpoints available in each direction). Anyway I've changed the default configuration setting to enable it.