IntergatedCircuits/USBDevice #17: HID Keyboard InEpNum value?

lukebayes, 2020-03-16:
As mentioned in another post, I'm working on getting a Composite USB device to represent itself as a CDC device to receive configuration commands, that will then cause it to also present as an HID keyboard, mouse or joystick.

As a caveat, I have lots of experience (> 2 decades) with a variety of programming languages, but C is not one of them, so I'm quite sure there are simple problems in what I'm doing here.

While attempting to get the keyboard HID device configured, I'm getting errors that look like they're related to my configuration of the keyboard keyboard interrupt address. Unfortunately, I can't seem to figure out what I'm doing wrong.

I found [this document](https://os.mbed.com/media/uploads/wim/hid_usb_intro_an249.pdf), which indicates the endpoint address format:
```
0x81, // bEndpointAddress
This item describes the address and data flow direction of the endpoint. Bits 0 through 3 define the endpoint’s address, and bit 7 describes the data flow direction, with 1 meaning IN and 0 meaning OUT. The item in this descriptor defines an “IN” endpoint with Endpoint Address 1.
```
I suspect there's a problem in the fact that I'm not generating nor returning a keyboard report, but I can't be certain because I can't seem to find an example of a valid keyboard report and I don't know how to author one.

Here is my usb_device.c configuration:
```c
#include "usb_device.h"

extern USBD_CDC_IfHandleType *const console_if;

// Keyboard HID interface
USBD_HID_IfHandleType _keyboard_if = {
        .App = &keyboard_app,
        .Base.AltCount = 1,
}, *const keyboard_if = &_keyboard_if;

// @brief USB device handle
USBD_HandleType hUsbDevice, *const UsbDevice = &hUsbDevice;

// @brief USB device configuration
const USBD_DescriptionType hdev_cfg = {
    .Vendor = {
        .Name           = "Eight Amps (8A)",
        .ID             = USBD_VID,
    },
    .Product = {
        .Name           = "Switchy & Infrareddy",
        .ID             = USBD_PID,
        .Version.bcd    = 0x0100 | HW_REV,
    },
#if (USBD_SERIAL_BCD_SIZE > 0)
    .SerialNumber       = (USBD_SerialNumberType*)DEVICE_ID_REG,
#endif
    .Config = {
        .Name           = "Switchy & Infrareddy",
        .MaxCurrent_mA  = 100,
        .RemoteWakeup   = 0,
        .SelfPowered    = 0,
    },
}, *const dev_cfg = &hdev_cfg;

void UsbDevice_Init(void) {
        KeyboardInit();

        // NOTE(lbayes): Made up this address, what value should it be?
        keyboard_if->Config.InEpNum = 0x89;
        // Mount the Keyboard interface
        USBD_HID_MountInterface(keyboard_if, UsbDevice);

        // All fields of Config have to be properly set up
        console_if->Config.InEpNum  = 0x81;
        console_if->Config.OutEpNum = 0x01;
        console_if->Config.NotEpNum = 0x8F;
        // console_if->Config.NotEpNum = 0x82;

        // Mount the CDC interface
        USBD_CDC_MountInterface(console_if, UsbDevice);

        // Initialize the device
        USBD_Init(UsbDevice, dev_cfg);

        // After charger detection the device connection can be made
        USBD_Connect(UsbDevice);
}

void UsbDevice_Deinit(void)
{
    USBD_Deinit(UsbDevice);
}
```

Here is my `/var/log/syslog` output when the device boots:
```bash
ar 16 13:30:16 beefcake kernel: [72877.892814] usb 3-4.2: USB disconnect, device number 101
Mar 16 13:30:21 beefcake kernel: [72882.978477] usb 3-4.2: new full-speed USB device number 102 using xhci_hcd
Mar 16 13:30:21 beefcake kernel: [72883.107182] usb 3-4.2: New USB device found, idVendor=0483, idProduct=0002, bcdDevice= 1.01
Mar 16 13:30:21 beefcake kernel: [72883.107185] usb 3-4.2: New USB device strings: Mfr=16, Product=32, SerialNumber=0
Mar 16 13:30:21 beefcake kernel: [72883.107186] usb 3-4.2: Product: Switchy & Infrareddy
Mar 16 13:30:21 beefcake kernel: [72883.107187] usb 3-4.2: Manufacturer: Eight Amps (8A)
Mar 16 13:30:21 beefcake kernel: [72883.115083] usbhid 3-4.2:1.0: couldn't find an input interrupt endpoint
Mar 16 13:30:21 beefcake kernel: [72883.118076] cdc_acm: probe of 3-4.2:1.1 failed with error -22
Mar 16 13:30:21 beefcake mtp-probe: checking bus 3, device 102: "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-4/3-4.2"
Mar 16 13:30:21 beefcake mtp-probe: bus: 3, device: 102 was not an MTP device
Mar 16 13:30:21 beefcake mtp-probe: checking bus 3, device 102: "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-4/3-4.2"
Mar 16 13:30:21 beefcake mtp-probe: bus: 3, device: 102 was not an MTP device
```

I suspect my problem is highlighted by the line with, `usbhid 3-4.2:1.0: couldn't find an input interrupt endpoint`.

When I run `sudo lshw -c input`, I get the following entry (others removed):
```bash
$ sudo lshw -c input

  *-usb:1 UNCLAIMED
       description: Keyboard
       product: Switchy & Infrareddy
       vendor: Eight Amps (8A)
       physical id: 2
       bus info: usb@3:4.2
       version: 1.01
       capabilities: usb-2.01
       configuration: maxpower=100mA speed=12Mbit/s
```
Here is my `keyboard_if.c` file:
```c
#include "keyboard_if.h"

void SetKeyboardReport(void* itf, USBD_HID_ReportType type, uint8_t * data, uint16_t length) {
        // What to do here?
}

void GetKeyboardReport(void* itf, USBD_HID_ReportType type, uint8_t reportId) {
        // What to do here?
}

void KeyboardInit(void) {
}

// .Init      = Init,
// .Deinit    = Deinit,
const USBD_HID_AppType keyboard_app = {
        .Name      = "Eight Amps (8A), Switchy",
        .SetReport = SetKeyboardReport,
        .GetReport = GetKeyboardReport,
};
```

Any tips or pointers are greatly appreciated and thanks again for such a great library.

IntergatedCircuits, 2020-03-16:
Hi!
In this peripheral silicon design the possible endpoint numbers/addresses are limited, only the first N can be used (check the reference manual of the exact type). The MSb of the address is reserved to indicate a Device -> Host direction (IN).
For example HID implementation you can refer to the [DebugDongle](https://github.com/IntergatedCircuits/DebugDongleFW/blob/master/Sensor/sens_if.c) interface(s). The HID report descriptor and the corresponding layout for standard keyboards is already defined in [this header](https://github.com/IntergatedCircuits/HidReportDef/blob/master/include/hid/keyboard.h).

HID uses interrupt endpoints, which have a specified polling interval, the host reads the endpoint with this frequency, but the device isn't required to send data in each interval. In practice the reports are only sent when a key state changes, and otherwise the device (the keyboard) gets into a low power mode on the bus.

lukebayes, 2020-03-17:
Thanks for the detailed and fast response(s)!

I did some initial work yesterday to pull this stuff down, but wasn't able to get it working (entirely due to my lack of experience). I need to focus on some other aspects of the product this week, but I'll get confirmation and detailed updates by next week.

lukebayes, 2020-04-07:
Just closing the issue as it's working now. Thanks!

closed 2020-04-07 by lukebayes