IntergatedCircuits/USBDevice #24: Windows 10 complains on Composite device configuration

lukebayes, 2020-07-27:
Attached a log output from USBView.exe when connecting a composite device:

[USBView-Aspen.txt](https://github.com/IntergatedCircuits/USBDevice/files/4983794/USBView-Aspen.txt)

Error suppressed after editing `Device/usbd_desc.c` with the following diff:

```diff
diff --git a/Device/usbd_desc.c b/Device/usbd_desc.c
index 9d8b82c..b6ab6ea 100644
--- a/Device/usbd_desc.c
+++ b/Device/usbd_desc.c
@@ -42,9 +42,9 @@ static const USB_DeviceDescType usbd_deviceDesc = {
     .bLength            = sizeof(USB_DeviceDescType),
     .bDescriptorType    = USB_DESC_TYPE_DEVICE,
     .bcdUSB             = USBD_SPEC_BCD,
-    .bDeviceClass       = 0x00,
-    .bDeviceSubClass    = 0x00,
-    .bDeviceProtocol    = 0x00,
+    .bDeviceClass       = 0xEF,
+    .bDeviceSubClass    = 0x02,
+    .bDeviceProtocol    = 0x01,
     .bMaxPacketSize     = USBD_EP0_MAX_PACKET_SIZE,
     .idVendor           = 0xFFFF,
     .idProduct          = 0xFFFF,
```

I'm pretty sure that's not a desirable change, but providing here as troubleshooting input...

Any tips appreciated.

benedekkupper, 2020-07-27:
Let me break this issue down. The USB 2.0 standard (Table 9-8.) states that

>**bDeviceClass** If this field is reset to zero, each interface within a configuration specifies its own class information and the various interfaces operate independently.

>**bDeviceProtocol** If this field is reset to zero, the device does not use class-specific protocols on a device basis. However, it may use class-specific protocols on an interface basis.

So in the case of composite USB devices in general, the device descriptor shouldn't specify class information, but should leave it to the individual interfaces.

The reason for the error message in usbview is that there's a somewhat contradicting requirement in *USB Interface Association Descriptor Device Class Code and Use Model* document (which is part of the USB 2.0 standard package):
> Devices that use the IAD must use the device class, subclass and protocol codes as defined in the example device descriptor illustrated in Table 1-1.

The motivation for this rule is to allow legacy host operating systems to bind the proper USB driver to the device, when they aren't prepared to interpret the IAD descriptor itself. The fact that the usbview is reporting this error actually means that it's meaningless, since it correctly recognizes the IAD, and displays this error as a kind of sanity check.

The reason I've ignored this error is because the precedence of these two rules isn't exactly clear, when only a subset of the device interfaces are linked together to form a single function, while the rest are expected to get a separate driver assigned. Performing the change that the error suggests actually creates a bigger problem on legacy OS-s: if they don't recognize these class codes, then none of the composite device's interfaces will work - which is a bigger issue than leaving the codes 0, when only the functions that use the IAD won't be working.

Can you tell me if there's anything about the USB device that isn't working as it should, but this change fixes it?

lukebayes, 2020-07-27:
After digging in a bit more, I don't think this was causing the problem we were having, it was just something I stumbled over as I was troubleshooting another issue.

Thanks for the detailed explanation, that makes sense to me, and I'll go ahead and close the issue.

closed 2020-07-27 by lukebayes