IntergatedCircuits/USBDevice #27: HID InData callback

alitekin2fx, 2021-06-02:
Is there any particular reason why the HID doesn't have an InData callback?

benedekkupper, 2021-06-02:
It's not added because it hasn't been a use-case so far, the HID applications I've worked on triggered the IN data transmission on external events. Does your application require the use of it?

alitekin2fx, 2021-06-02:
Yes, I try to use your library in my CMSIS DAP project. It needs IN and OUT endpoints.

benedekkupper, 2021-06-02:
OK, I'm adding this request to my work queue, ETC 1 week. If it's more urgent, you can do it yourself as well, just add an `InData` function to `hid_cbks` and redirect it to the application callback.

benedekkupper, 2021-06-05:
Hi Ali, please try this master now, and close the ticket if it satisfies your requirements.

closed 2021-06-08 by alitekin2fx