IntergatedCircuits/USBDevice #32: Device enumeration fails for composite device of NCM and two HID

FelixGruetzmacher, 2022-03-30:
We are observing that Windows and iOS both fail to enumerate our device. Windows reports a “Device not migrated due to partial or ambiguous match” message.
Initialization is given as:
    /* All fields of Config have to be properly set up */
    hid_if.Config.InEpNum  = 0x83;
    hid_if.Config.OutEpNum = 0x03;
    hid_if.App = &hid_app;
    keyboard_if.Config.InEpNum  = 0x84;
    keyboard_if.Config.OutEpNum = 0x04;
    keyboard_if.App = &keyboard_app;
    ncm_usb_if->Config.InEpNum = 0x81;
    ncm_usb_if->Config.OutEpNum = 0x01;
    ncm_usb_if->Config.NotEpNum = 0x82;
    /* Mount the interfaces to the device */
    USBD_NCM_MountInterface(ncm_usb_if, UsbDevice1);
    USBD_HID_MountInterface(&hid_if, UsbDevice1);
    USBD_HID_MountInterface(&keyboard_if, UsbDevice1);

And here is my configuration:
#define USBD_MAX_IF_COUNT           8
#define USBD_EP0_BUFFER_SIZE        512
#define USBD_HS_SUPPORT             0
#define USBD_SERIAL_BCD_SIZE        0
#define USBD_MS_OS_DESC_VERSION     2
#define USBD_CDC_NOTEP_USED         0
#define USBD_CDC_CONTROL_LINE_USED  0
#define USBD_CDC_BREAK_SUPPORT      0
#define USBD_DFU_ALTSETTINGS        0
#define USBD_DFU_ST_EXTENSION       0
#define USBD_HID_ALTSETTINGS        0
#define USBD_HID_OUT_SUPPORT        1
#define USBD_HID_REPORT_STRINGS     0

It is important to note that as soon as we drop any of the three components, the device is successfully enumerated. Only with all three present do we encounter this problem.
Could you please advise?

benedekkupper, 2022-03-30:
Can you send me
1. Screenshot of what Windows Device Manager shows?
2. Screenshot of the USB descriptors of your device? E.g. using [usbview](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview)
3. USB traces maybe?

Note that the HID class currently doesn't have a configuration option for indicating 101 keyboard (or mouse) support, for that you would need to patch the code yourself in `hid_getDesc()` so that `bInterfaceSubClass` and `bInterfaceProtocol` are set correctly (see comments in usbd_hid.c).

Experiment with different USB Vendor and Product IDs, maybe the OS has cached some drivers for your IDs from your previous attempts, and rejects the new descriptor.

benedekkupper, 2022-03-31:
The usbview result has this line which should be alarming:
> Microsoft OS Descriptor is not available. Error code: 0x0000001F

So please check if you still get this error when the enumeration succeeds, and try to get closer to the root cause.

You can use WireShark with usbpcap to capture USB communication.

FelixGruetzmacher, 2022-04-01:
Important new insight: When we mount the interfaces on the HS port rather than the FS port of our STM32F429 eval board, all three interfaces are working fine.
The error in retrieving the Microsoft OS descriptor, however, is still present. Do you know a strategy for tracking down why this particular descriptor might not be available even though our configuration clearly says it should be present? It's not much of an issue for us but I'd like to understand what's happening.