IntergatedCircuits/USBDevice #33: ncm:Out in itf

momoxuelin, 2022-07-13:
Hi,I'm a newbie，I want to use ncm in my project, but I don't know why there are two Out page in itf?

momoxuelin, 2022-07-13:
Or is there any documentation to help me understand the source code?

benedekkupper, 2022-07-13:
The comment says "Received NTB status and double buffer" which is really all you need to know. You should look at the USBD_NCM_Exported_Functions section of functions, which is the public API of the class. For a demonstration project on how NCM can be integrated and used, see [IPoverUSB](https://github.com/IntergatedCircuits/IPoverUSB) project. If you want to understand the internal operation of the source code, you can read the comments in the file, and read the USB NCM class specification.

momoxuelin, 2022-07-13:
OK, thank you very much.

closed 2022-07-17 by benedekkupper