IntergatedCircuits/USBDevice #35: USB CDC data issues when no USB hub is present

Mogkal, 2024-09-30:
Hello
I am having and issue where the USB data seems to be corrupt when there is no USB hub on the USB connection.
I have been using your USB device library for quite some time (Nice work btw) without any issues, mainly for a composite CDC+HID device. I am running on a custom PCB with an STM32L4, and so far there has always been a USB hub on the PCB or on the PC that it is connected to (note that most PC's have a USB hub on the USB ports). Recently however, there hasn't been any need for a USB hub anymore, so it has been removed from the PCB. Now without the HUB there seems to be some data corruption and i am currently trying to troubleshoot whether it is a hardware/firmware issues, and whether it is my code, or any libraries that i am using. The MCU is running FreeRTOS and the data is sent to a desktop application. The data has varying length (200-1500 bytes), and has a CRC at the end of each packet. Without the hub i am getting CRC errors consistently, but when a hub is connected in some way there are no CRC errors.

Do you have any ideas? maybe it has been fixed in the newer library version? Any help is much appreciated.

Mogkal, 2024-10-01:
After some more debugging, it seems to be related to the USB LPM (Link power management). When LPM is enabled and no USB hub is present, the USB link is in sleep state, and doesnt wake up. But when a hub is present, the USB link is in idle mode whenever data is not sent.

benedekkupper, 2024-10-15:
When having the LPM bit enabled in the firmware, and no USB hub between the device and the host, the OS tries to use LPM and as a result, the MCU USB link stays in sleep mode. There is an explanation from microsoft that says the LPM is only used on a root USB port, and not on a USB hub downstream port: https://techcommunity.microsoft.com/t5/microsoft-usb-blog/link-power-management-lpm-in-usb-2-0/ba-p/270812

The issue is mainly because of enabling LPM without handling it properly to make sure the MCU supports it.

closed 2024-10-15 by benedekkupper